dhylands/python_lcd #13: Unable to print on LCD

nikhil1983, 2017-12-29:
Hello,
I am a newbie in Micropython. I am working on **STM32F407Discovery** board. I wanted to display character on LCD (16x2) display. I tried to use the module given by you, but its not displaying any character on LCD.

Below is code ..

> from pyb import Pin, delay, millis
> from pyb_gpio_lcd import GpioLcd
> 
> #Configuring LCD Pins
> rs = pyb.Pin('PB1',Pin.OUT_PP,Pin.PULL_UP)
> rw = pyb.Pin('PB4', Pin.OUT_PP,Pin.PULL_UP)
> en = pyb.Pin('PB5', Pin.OUT_PP,Pin.PULL_UP)
> d0 = pyb.Pin('PE8', Pin.OUT_PP,Pin.PULL_DOWN)
> d1 = pyb.Pin('PE9', Pin.OUT_PP,Pin.PULL_DOWN) 
> d2 = pyb.Pin('PE10', Pin.OUT_PP,Pin.PULL_DOWN)
> d3 = pyb.Pin('PE11', Pin.OUT_PP,Pin.PULL_DOWN)
> d4 = pyb.Pin('PE12', Pin.OUT_PP,Pin.PULL_DOWN)
> d5 = pyb.Pin('PE13', Pin.OUT_PP,Pin.PULL_DOWN)
> d6 = pyb.Pin('PE14', Pin.OUT_PP,Pin.PULL_DOWN)
> d7 = pyb.Pin('PE15', Pin.OUT_PP,Pin.PULL_DOWN)
> 
> def test_main():
>     """Test function for verifying basic functionality."""
>     print("Running test_main")
>     lcd = GpioLcd(rs,en,d0,d1,d2,d3,d4,d5,d6,d7,2,16)
>     lcd.putstr("It Works!\nSecond Line")
>     delay(3000)
>     lcd.clear()
>     count = 0
>     while True:
>         lcd.move_to(0, 0)
>         lcd.putstr("%7d" % (millis() // 1000))
>         delay(1000)
>         count += 1

I have initalised the LCD pins correctly and the code is getting executed properly, still i am unable to understand why nothing gets printed on LCD. Can anyone help me in the above code and how to implement on STM32F407disc board.

Hoping for some quick help..

nikhil1983, 2018-05-09:
Hey, 
Yes, LCD is working properly, as rows of black square appear when i turn LCD on. But still haven't found any solution to use this module. 

I have used async method, published by Sir Peter Hinch and it works well. So sir, if you have discovery board can you try out this module with it.

dhylands, 2018-05-10:
ok - I found a few errors in your script.

1 - You're missing an `import pyb` or yopu should be using Pin instead of pyb.Pin
2 - You forgot to pass in the rw_pin and backlight_pin parameters when you call GpioLcd (this is why the example uses named arguments)
3 - There is no need to initialize the output pins. GpioLcd does this already. You can just pass in pyb.Pin('PB1') or pyb.Pin.board.PB1
4 - PULL_UP/PULL_DOWN only apply to input pins

The last 2 points aren't really errors per say. Here's a modified version of your script that works:
```
from pyb import Pin, delay, millis
from pyb_gpio_lcd import GpioLcd

#Configuring LCD Pins
rs = Pin('PB1')
rw = Pin('PB4')
en = Pin('PB5')
d0 = Pin('PE8')
d1 = Pin('PE9')
d2 = Pin('PE10')
d3 = Pin('PE11')
d4 = Pin('PE12')
d5 = Pin('PE13')
d6 = Pin('PE14')
d7 = Pin('PE15')

def test_main():
  """Test function for verifying basic functionality."""
  print("Running test_main")
  lcd = GpioLcd(rs,en,d0,d1,d2,d3,d4,d5,d6,d7,rw,None,2,16)
  lcd.putstr("It Works!\nSecond Line")
  delay(3000)
  lcd.clear()
  count = 0
  while True:
    lcd.move_to(0, 0)
    lcd.putstr("%7d" % (millis() // 1000))
    delay(1000)
    count += 1
```

nikhil1983, 2018-05-11:
Hello thanks, its working. just wanted to know how can we shift our message right or left on LCD using this module ?

dhylands, 2018-05-11:
The HD44780 supports a MOVE display/cursor left./right command. I don't have a wrapper for it in LcdApi, but I was able to get it to scroll using:
```
      lcd.hal_write_command(lcd.LCD_MOVE | lcd.LCD_MOVE_DISP)
```
or
```
      lcd.hal_write_command(lcd.LCD_MOVE | lcd.LCD_MOVE_DISP | lcd.LCD_MOVE_RIGHT)
```

nikhil1983, 2018-05-14:
But the issue with above command are once the character are shifted out from right, the shifted out character are not getting shifted in from left. 

So if i want to make a scrolling display, "ABCD" once the D goes out of 16th column, in the next loop D Should come in 1st column and C in 16th Column.

dhylands, 2018-05-14:
Yep - that's the same behaviour I see as well.

The hardware appears to be inserting spaces, which means that you'd need to use the cursor positioning commands and draw the character that you want shifted in

dhylands, 2018-05-22:
I suspect that you must have some delays in your code. I can shift in 10 characters in 10 milliseconds.

According to the HD47780 datasheet, when you shift the display, you're actually shifting the DDRAM address. This means that after shifting the display by 1 character right, then the top-left corner is at x=0x27. See https://www.sparkfun.com/datasheets/LCD/HD44780.pdf page 12 for details.

Here's the sample code I tested with (using the 8-bit parallel GPIO on the STM32F407 Discovery board):
```python
import pyb

def test_main():
    """Test function for verifying basic functionality."""
    print("Running test_main a")
    rs = pyb.Pin('PB1',Pin.OUT_PP)
    rw = pyb.Pin('PB4', Pin.OUT_PP)
    en = pyb.Pin('PB5', Pin.OUT_PP)
    d0 = pyb.Pin('PE8', Pin.OUT_PP)
    d1 = pyb.Pin('PE9', Pin.OUT_PP)
    d2 = pyb.Pin('PE10', Pin.OUT_PP)
    d3 = pyb.Pin('PE11', Pin.OUT_PP)
    d4 = pyb.Pin('PE12', Pin.OUT_PP)
    d5 = pyb.Pin('PE13', Pin.OUT_PP)
    d6 = pyb.Pin('PE14', Pin.OUT_PP)
    d7 = pyb.Pin('PE15', Pin.OUT_PP)
    rw.value(0)
    lcd = GpioLcd(rs_pin=rs,
                  enable_pin=en,
                  d0_pin=d0,
                  d1_pin=d1,
                  d2_pin=d2,
                  d3_pin=d3,
                  d4_pin=d4,
                  d5_pin=d5,
                  d6_pin=d6,
                  d7_pin=d7,
                  num_lines=2, num_columns=16)
    lcd.putstr("It Works!\nSecond Line\nThird Line\nFourth Line")
    t1 = pyb.millis()
    for i in range(10):
      lcd.move_to(0x27 - i, 0)
      lcd.hal_write_data(ord('0') + i)
      lcd.hal_write_command(lcd.LCD_MOVE | lcd.LCD_MOVE_DISP | lcd.LCD_MOVE_RIGHT)
      #delay(1000)
    t2 = pyb.millis()
    print('Delta =', t2 - t1, 'msecs')
    pyb.delay(1000)
    lcd.clear()
    count = 0
    while True:
        lcd.move_to(0, 0)
        lcd.putstr("%7d" % (millis() // 1000))
        delay(1000)
        count += 1

test_main()
```