dhylands/python_lcd #19: Putting this code on my Nodemcu V3

pmulvey, 2018-07-28:
I am a newbie and am running EsPy V1.0.07 on Win 8 with a Nodemcu V3. After a long struggle with moving around import statements I finally got a library for the PCF8574 working. I am now trying your I2C LCD stuff but am unsure as to how to get the code to run. When I run esp8266_i2C_lcd_test.py I get:
```
Running test_main
Traceback (most recent call last):
  File "<stdin>", line 47, in <module>
  File "<stdin>", line 15, in test_main
  File "esp8266_i2c_lcd.py", line 26, in __init__
OSError: [Errno 19] ENODEV
```
I know that it must be some newbie issue so I would appreciate your help.

Regards
Paul

metoikos, 2018-08-12:
@pmulvey if you following this answer => https://forum.micropython.org/viewtopic.php?t=2858#p16925,
try to change 0x27 to 0x3F which some lcd's requires this address.

Full line:
`lcd = I2cLcd(i2c, 0x3F, 2, 16)`

dhylands, 2018-08-20:
ENODEV means that the device wasn't found on the i2c bus. You need to be able to see the device on the bus before the library will work (as @metoikos alludes to). You should also try using the i2c scan method to see what devices are reported. keep in mind that the reported addresses are in decimal (i.e. base 10).