dhylands/python_lcd #21: Need driver for chip SPLC780D1

swanduron, 2018-12-16:
Sorry for borther.
I want to use this function to drive LCM4004 module (based on daul SPLC780D1 chip) but it seems does not work as expected. For example, sometimes I got some wrong chars on screen and generally no changes on screen.
The LCM4002 module also use this chip, could help me to give a look for this issue?

Document link:
http://www.vatronix.com/upload/LCD_Controller/splc780d1.pdf
Ebay link:
https://www.ebay.com/itm/404-40X4-4004-Character-LCD-Module-Display-Screen-LCM-SPLC780D-Controller/202481696942?hash=item2f24d980ae:m:m0y8ZYK7spyvF9zJLMZG6DQ:rk:1:pf:0

dhylands, 2018-12-16:
You'll need to compare the command set with the HD77480 which is what this library is designed to drive.
Read this thread: https://www.eevblog.com/forum/beginners/lcd-hd44780-vs-splc780d-whats-the-difference-good-large-lcd/ 
It seems that some of the timing characteristics might be different.

swanduron, 2018-12-17:
Thanks master, I will check this feature tonight~

swanduron, 2018-12-17:
Very sorry for my mistake. The LCM4004 module has 2 different pin map: LCM4004 uses pin 1 as DB7 but LCM4004A uses pin 1 as DB0, modules on my hand are LCM4004.

Your code works very beautiful in this chip, thanks for your help.

closed 2018-12-17 by swanduron