dhylands/python_lcd #24: Unable to get 16x2 LCD working with STM32F401RE Nucleo

rlnmadbee, 2019-05-31:
Hi Dave, I have been trying to implement your pyb_gpio_lcd.py for a 16x2 lcd on micropython with no success.

I am using stm32f401re nucleo. 
4-bit mode (d4,d5,d6,d7)
rw has been grounded
rs and en have been tied to PC14 and PC15 respectively
PC2, PC3, PC4 and PC5 has been used for data line. 
I have connected a 10K pot for contrast adjustment . all i could see is black boxes on the first row.

Please advise what am I missing.

dhylands, 2019-06-06:
Looking at the schematics, {PC14 and PC15 are connected to the 32 kHz oscillator which is probably the main problem.

I connected it up by grounding the contrast pin (a quick hack - using the potentiometer is better) and connecting RS and EN to PC6 and PC8 instead.

I put some photos here: https://photos.app.goo.gl/81BKYmFBVqahvCnr7

I modified pyb_gpio_lcd_test.py and called it lcd_401.py and put lcd_401.py, lcd_api.py and pyb_gpio_lcd.py in the ports/stm32/modules directory (so they'd be built as frozen modules).

I changed lcd_401.py so that the call to GpioLcd looked like this:
```
    lcd = GpioLcd(rs_pin=Pin.cpu.C6,
                  enable_pin=Pin.cpu.C8,
                  d4_pin=Pin.cpu.C2,
                  d5_pin=Pin.cpu.C3,
                  d6_pin=Pin.cpu.C4,
                  d7_pin=Pin.cpu.C5,
                  num_lines=2, num_columns=16)
```
I changed the pin assignments and the num_lines and num_columns.

Here's what I ran from the REPL on the NUCLEO_F401RE to get the test to work.
```
MicroPython v1.11-37-g62f004ba4 on 2019-06-06; NUCLEO-F401RE with STM32F401xE
Type "help()" for more information.
>>> import lcd_401
>>> lcd_401.test_main()
Running test_main
```

rlnmadbee, 2019-06-08:
Dave, you are always complementing!!

Thanks again for the wonderful reply there!! Have got it working. Didn't notice that those pins were connected to oscillator. 

Thank you so much for your time and effort in setting up and posting that picture.

closed 2019-06-08 by rlnmadbee