dhylands/python_lcd #25: Unable to print in LCD - GPIO

JesusTJ, 2019-06-15:
Hello , i am trying to learn micropython and I got the board **NEW NodeMcu Lua ESP8266 CH340G ESP-12E Wireless WIFI Internet Development** Board from ebay.

I already completed some practices and now i'm trying to display information in LCD 1602A with GPIO pins but can not. only black squares.

i loaded esp8266-20190529-v1.11 With NodeMCU-PyFlasher-4.0-x64

blob:https://web.whatsapp.com/883eb399-9f53-4f52-8023-40e25437819a
[image]

Can i get some help ?

dhylands, 2019-06-19:
In order to help, I'd need to be able to see your code, see the output you get from running the code and you'd need to share how you wired up the LCD.

JesusTJ, 2019-06-25:
Hello , If i use Arduino it works with the following code :

#include <LiquidCrystal.h>
#include "DHT.h"

//LiquidCrystal lcd(RS, EN, d4, d5, d6, d7);
// RS = D2, EN = D3, d4 = D5, d5 = D6, d6 = D7, d7 = D8;

LiquidCrystal lcd(4, 0, 14, 12, 13, 15);
byte lcdRows            = 2;
byte lcdCols            = 16;

#define DHTPIN 5
#define DHTTYPE DHT22

DHT dht(DHTPIN , DHTTYPE);

int timeSinceLastRead = 0;

void setup() {
  Serial.begin(115200);
  delay(10);
  lcd.begin(lcdCols, lcdRows);
  lcd.clear();
  lcd.setCursor(0, 0);
  lcd.print("  STARTING UP    ");
  lcd.setCursor(0, 1);
  lcd.print(" Temp & Hum ");
  dht.begin();
}

void loop() {
    delay(500);
    if(timeSinceLastRead > 2000)
    {
      float h = dht.readHumidity();
      float t = dht.readTemperature();
      float f = dht.readTemperature(true);

      lcd.clear();
      lcd.setCursor(0, 0);
      lcd.print(" Temp :");      
      lcd.print(t);
      lcd.setCursor(0, 1);
      lcd.print(" Hum  :");      
      lcd.print(h);

      if (isnan(h) || isnan(t) || isnan(f))
      {
        Serial.println("Failed to read from DHT sensor!");
        timeSinceLastRead = 0;
        return;
      }
      timeSinceLastRead = 0;
    }
    timeSinceLastRead += 100;  
}

[image]

I want to do the same but with python

Can you share with me an example ?

dhylands, 2019-06-26:
The code is known to work. The most likely problem is mis-wiring or the code and the wiring not agreeing with each other. Unfortunately, I don't have any NodeMcu's.  Here's a recent example where I connected up to a NUCLE_F401RE: https://github.com/dhylands/python_lcd/issues/24

dhylands, 2019-06-26:
I'm not a 100% sure, but it may also be that the Pin numbers being used by Arduino aren't the same as the pin numbers being used by MicroPython.

dhylands, 2019-07-03:
I bought a NodeMCU v3 and it seems to be working. I wired it up as follows:

LCD pin1 - NodeMCU Ground
LCD pin 2 - NodeMCU VV (VUSB)
LCD pin 3 - Contrast - NodeMCU Ground
LCD pin 4 - RS - NodeMCU D2 - GPIO 4
LCD pin 5 RW - NodeMCU Ground
LCD pin 6 EN - NodeMCU D3 - GPIO 0
LCD pin 7, 8, 9, 10 - unconnected
LCD pin 11 - D4 - NodeMCU D5 - GPIO 14
LCD pin 12 - D5 - NodeMCU D6 - GPIO 12
LCD pin 13 - D6 - NodeMCU D7 - GPIO 13
LCD pin 14 - D7 - NodeMCU D4 - GPIO 2

I initially tried connecting LCD pin 14 to NodeMCU pin D8, put for some reason my REPL stopped working when I did that, so I used D4 instead.

I edited nodemcu_gpio_lcd_test.py to use the pins numbers I mentioned above:
```
    lcd = GpioLcd(rs_pin=Pin(4),
                  enable_pin=Pin(0),
                  d4_pin=Pin(14),
                  d5_pin=Pin(12),
                  d6_pin=Pin(13),
                  d7_pin=Pin(2),
                  num_lines=2, num_columns=20)
```
I copied nodemcu_gpio_lcd_test.py, nodemcu_gpio_lcd.py, and lcd_api.py to the esp8266 filesystem, and ran the following:
```
MicroPython v1.11-8-g48dcbbe60 on 2019-05-29; ESP module with ESP8266
Type "help()" for more information.
>>> 
>>> import nodemcu_gpio_lcd_test
>>> nodemcu_gpio_lcd_test.test_main()
Running test_main
```
and got this output:
[image]

JesusTJ, 2019-07-04:
Something is wrong , is not working , only black squares

I followed your recomendations
[image]

purpleidea, 2019-07-04:
On Wed, Jul 3, 2019 at 10:06 PM JesusTJ <notifications@github.com> wrote:
>
> Something is wrong , is not working , only black squares

Make sure the contrast is *exactly right*. Some screens don't show the
chars unless this is tuned perfectly.

dhylands, 2019-07-04:
If you can see the black squares then you should be able to see the text (as far as contrast goes).

That tells me that something is wrong with the wiring. Maybe one of the wires has a broken contact, or the wiring doesn't agree with what's being initialized.

I assume that you're running test_main() after doing the import (the import by itself won't change the display)