dhylands/python_lcd #27: 16x4 LCD have 3rd and 4th line shifted by 4 chars

petrkr, 2019-09-23:
When you using 16x4 (instead 20x4) display, then due this line buffer is shifted by 4 chars

https://github.com/dhylands/python_lcd/blob/87ca81a7da025b9f60d857c902435bc87dcf842f/lcd/lcd_api.py#L131

when you change this 0x14 to 0x10, it works. I suggest somehow IF it based on LCD_COLS

somethings like 0x10 if  LCD_COLS==16 else 0x14

dhylands, 2019-09-30:
0x14 == 20, so it could probably just do addr += self.num_columns

petrkr, 2019-09-30:
Maybe it could works for all situations... Here is that page which I wanna sent in first post, but took me 7 days to google it (shame me I do not use bookmarks)

http://web.alfredstate.edu/faculty/weimandn/lcd/lcd_addressing/lcd_addressing_index.html

dhylands, 2019-09-30:
Heh - I actually found that page just before answering. So it looks like it should work fine. I don't have any 16 x 4 LCD panels (just some 20 x 4 LCD panels. So if you could verify that the fix works for you, I'll verify it works on my 20 x 4.

petrkr, 2019-09-30:
I should have both 16x4 and 20x4 aswell..  So I can test it.. For now I did easily fix https://github.com/octopusengine/octopuslab/commit/d3d98b1cf063dfaea2b4456f6815365fe226f47a

But I will try do that += mentioned before... Whenever I will get time :)

Petr