dhylands/python_lcd #32: add seeed studio Grove - 16x2 LCD support - any chance?

dsssssssss9, 2021-05-06:
Hi

I currently have a few of these ....

https://wiki.seeedstudio.com/Grove-16x2_LCD_Series/

This seems to be based on a JDH1804  controller....

https://github.com/SeeedDocument/Grove-16x2_LCD_Series/raw/master/res/JDH_1804_Datasheet.pdf

i am guessing that this is why i cannot seem to make it work under circuit python with your wonderful library?

Is there any way you could add support for this controller please?

I hope you don't mind me asking?

Regards

Jason

davehylands, 2021-05-07:
The LCD instruction set looks very similar (if not identical). It looks like it basically has a built-in I2C/parallel conversion.

Whereas the existing drivers are all assuming an external i2c-gpio expander chip. This LCD will need a different HAL layer.

The datasheet doesn't give any examples of sending i2c commands to the device, so you'd probably have to figure out what's happening by looking at their arduino code.

davehylands, 2021-05-07:
I noticed that there is a pyb_i2c_grove_rgb_lcd.py (the HAL layer) and pyb_i2c_grove_rgb_lcd_test.py (a test program), and I found a Grove-LCD RGB backlight in my grove parts. I was able to hook it up to my pyboard and get it to work. I had to power the Grove board using 5V (the backlight worked with 3.3v but not the LCD itself).

I noticed that the contrast was little poor powering the LCD using V+ on the pyboard. Apparently the LCD really wants 5v and V+ is only 4.3v. When I powered it using VBUS (which I measured at 5.1v) then the contrast was much better.

dsssssssss9, 2021-05-09:
sorry for the delay in replying!

I will give this a go next few days 

appreciate the help!