dhylands/python_lcd #44: Could you please add a new class to i2c_lcd.py?

FlewittGuy, 2022-12-21:
**Request**
So while using your library I created a simple class which I added to the `/lcd/i2c_lcd.py` file. The only purpose is to organize the main file by using one library instead of two. I am here to ask if you could add the class below to the bottom of the `/lcd/i2c_lcd.py`  file. Thanks.

**Class**
```py
class I2C_LCD():
    def __init__(self, SDA_PIN, SCL_PIN, ROWS, COLS, FREQ):
        self.SDA_PIN = SDA_PIN
        self.SCL_PIN = SCL_PIN
        
        self.I2C_NUM_ROWS = ROWS
        self.I2C_NUM_COLS = COLS
        
        self.FREQ = FREQ
        
        self.i2c = SoftI2C(sda=Pin(self.SDA_PIN), scl=Pin(self.SCL_PIN), freq=self.FREQ)
        self.I2C_ADDR = int(hex(self.i2c.scan()[0]))  # Finds the unique 7-bit I2C address
        self.lcd = I2cLcd(self.i2c, self.I2C_ADDR, self.I2C_NUM_ROWS, self.I2C_NUM_COLS)
```

**Example/Main.py File**
```py
from i2c_lcd import I2C_LCD

disp1 = I2C_LCD(
    SDA_PIN = 0,
    SCL_PIN = 1,
    ROWS    = 2,
    COLS    = 16,
    FREQ    = 400_000
)

disp2 = I2C_LCD(
    SDA_PIN = 4,
    SCL_PIN = 5,
    ROWS    = 2,
    COLS    = 16,
    FREQ    = 400_000
)

disp1.lcd.putstr(f"Hello, World!")
disp2.lcd.putstr(f"Hello, Again!")
```

FlewittGuy, 2022-12-21:
Its just a request, I know I can just fork it but I wanted to shoot my shot.

dhylands, 2023-01-12:
So this looks like a totally independent class which should probably go into its own file?

The file i2c_lcd.py is intended to be used on a linux (like a Raspberry Pi or a BeagleBne Black). Your class looks to be Micropython specific. So it doesn't make sense to mix these 2 things.