dhylands/python_lcd #46: Reduce code size and memory footprint

rkompass, 2023-01-09:
On the basis of the code here which I gratefully took as a starting point I derived a driver (I2C) which has reduced code size and memory footprint, while somewhat improving functionality. It is [here](https://github.com/rkompass/LCD_HD44780_PCF8574_mPy).
Memory consumption went down from 4896 to 1616 bytes.
This was achieved by combining functions and replacing constant variables with their values.
The new version also should be easier for new users, as it is only one file, which probably should run on many platforms.
I would be happy to provide a set of  (or a combined) pull request, if you are interested (thx for feedback here).
The same for a version for other interfaces than I2C and a more universal test script.

dhylands, 2023-01-12:
I'd definitely be interested in taking a look.

dhylands, 2023-01-12:
@rkompass Unfortunately the link above doesn't seem to be working

rkompass, 2023-01-12:
Hello Dave,
thank you, I see, there went something wrong. I'm not experienced yet with Github.

The file lcd_i2c8574.py (my driver) is this:
```
# Implements a HD44780 character LCD connected via PCF8574 on I2C
# Should work with ESP8266 and (tested) Pyboard 1.1
#
# Derived from https://github.com/dhylands/python_lcd by rkompass 2022
#    (put everything in one file, remove constants, simplify....) resulting in only ~1.6 K memory consumption.

from time import sleep_ms, sleep_us

class I2cLcd: # Implements a HD44780 character LCD connected via PCF8574 on I2C

    def __init__(self, i2c, i2c_addr, n_lin, n_col):
        self.i2c = i2c
        self.i2c_addr = i2c_addr
        self.n_lin = min(n_lin, 4)
        self.n_col = min(n_col, 40)
        self.cur_x = 0
        self.cur_y = 0
        self.impl_nl = False         # implicit newline: we have just wrapped around to start of new line
        self.backl = 0x08

        self.i2c.writeto(self.i2c_addr, bytearray([0]))          # init I2C
        sleep_ms(20)                                             # Allow LCD time to powerup
        for _ in range(3):                                       # send reset 3 times
            self.i2c.writeto(self.i2c_addr, bytearray((0x34, 0x30))) # LCD_FUNCTION_RESET
            sleep_ms(5)                                          # need to delay at least 4.1 msec
        self.i2c.writeto(self.i2c_addr, bytearray((0x24, 0x20))) # LCD_FUNCTION, put LCD into 4 bit mode
        sleep_ms(1)
        self.set_display(False)
        self.clear()
        self._wr(0x06)                         # LCD_ENTRY_MODE | LCD_ENTRY_INC
        self.set_cursor(False)
        self.set_display(True)                 # we might include a backlight option here
        self._wr(0x28 if n_lin > 1 else 0x20)  # LCD_FUNCTION_2LINES if n_lin > 1 else LCD_FUNCTION

    def clear(self):        # Clears the LCD display and moves the cursor to the top left
        self._wr(0x01)      # LCD_CLR
        self._wr(0x02)      # LCD_HOME
        self.cur_x = 0
        self.cur_y = 0

    def set_cursor(self, show=False, blink=False): # Causes the cursor to be made visible if show or even blink
        self._wr(0x0f if blink else (0x0e if show else 0x0c))  # LCD_ON_CTRL | LCD_ON_DISPLAY | (LCD_ON_CURSOR) 

    def set_display(self, on=True, backl=None):  # Turns the LCD on (unblanks) or off, optionally sets backlight
        self._wr(0x0c if on else 0x08)  # LCD_ON_CTRL | LCD_ON_DISPLAY
        if backl is not None:
            self.backl = 0x08 if backl else 0x00
            self.i2c.writeto(self.i2c_addr, bytearray((self.backl,)))

    def move_to(self, cur_x, cur_y):  # Moves the cursor position to the indicated position
        self.cur_x = cur_x            # The cursor position is zero based (x == 0 -> first column)
        self.cur_y = cur_y
        addr = cur_x & 0x3f
        if cur_y & 1:     # Lines 1 & 3 add 0x40
            addr += 0x40 
        if cur_y & 2:    # Lines 2 & 3 add number of columns
            addr += self.n_col
        self._wr(0x80 | addr) # LCD_DDRAM | ..

    def write(self, string='', end='\n'):  # Writes the string at the current cursor pos and advances cursor
        for c in ''.join((string, end)):
            if self.cur_x == 0:                   # Clear the line that we have just begun.
                for _ in range(self.n_col):
                    self._wr(32, 1)               # ord(' ')  = 32
                self.move_to(0, self.cur_y)
[... 102 lines cut ...]
    lcd.write(chr(i), end='')
    sleep_ms(200)

sleep_ms(1000)
lcd.clear()

print("LCD write at (0,0)-position")
lcd.move_to(0, 0)
lcd.write("Here from (0,0) on")
sleep_ms(2000)
print("LCD write at (5,3)-position")
lcd.move_to(5, 3)
lcd.write("Here from (5,3) on")
sleep_ms(2000)
lcd.clear()
lcd.write("Again a line, which will be quite long.")
lcd.write("Short line")
lcd.write("X")
lcd.write()
lcd.write('Short', end='')
lcd.write('..short', end='')
lcd.write('..fine!')
```

Thank you for comments. Raul

freemansoft, 2023-01-15:
The file doesn't appear to be in the referenced repository.

rkompass, 2023-01-20:
For that reason I presented the file above. I now found a bug in write() and will try to remove it in the next days (without hurry). After that I will setup the repository again.

rkompass, 2023-02-22:
@dhylands @freemansoft 
I have updated my code and it is now in the Github Repository [here](https://github.com/rkompass/Lcd_I2c8574).